ros/dynamic_reconfigure #51: Unable to reload parameters from a file generated by dumping

b1willaert, 2015-02-10:
I am using ubuntu 14.04/Indigo from debs. 

I am not using grouping. The generated yaml file contains all parameters _twice_ as follows:

!!python/object/new:dynamic_reconfigure.encoding.Config
dictitems:
  parameter1: false
...
  groups: !!python/object/new:dynamic_reconfigure.encoding.Config
    dictitems:
      parameter1: false
...
      groups: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      id: 0
      parameters: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      parent: 0
      state: true
      type: ''
    state: []
state: []
 

The exact same issue has been posted on the ROS.answers too:
http://answers.ros.org/question/181915/dynamic_reconfigure-unable-to-reload-dumped-parameters/

esteve, 2015-02-14:
Does this only happen with nodelets or with nodes as well?

b1willaert, 2015-02-17:
I only tested it with nodes. So yes, it happens with nodes.

pirobot, 2015-03-15:
I am experiencing the same issue, both under ROS Hydro/Ubuntu 12.04 *and* ROS Indigo/Ubuntu 14.04.  The following error appears when dumping parameters to a file for either a node or nodelet and then trying to load them immediately after:

    $ rosrun dynamic_reconfigure dynparam dump /my_node my_node.yaml
    $ rosrun dynamic_reconfigure dynparam load /my_node my_node.yaml

    error updating parameters: don't know parameter: groups

In addition to the error, the parameters are *not* loaded (just to be clear).

l0g1x, 2015-10-09:
Has anyone looked into this issue lately? It would be useful to re-load things like path planner parameters from adjusting values in rqt_reconfigure. param server names may be different from dynamic reconfigure .cfg variable names in some packages, so that might be a issue if one wants to load the dumped dynamic reconfigure yaml to the ros param server, for a quick restart where they left off.

l0g1x, 2015-12-09:
Can this one be added to the bug sqaush list? I can look into this but not until friday

esteve, 2015-12-09:
@l0g1x sure, thanks for volunteering!

esteve, 2015-12-09:
@l0g1x I just added it to the document, please claim it so that we can coordinate better.

https://docs.google.com/document/d/1DPZuBDyUyOVNkKCOtwpo-nxGba7aSJdvP6ncqRZOBEU/edit?usp=sharing

l0g1x, 2015-12-09:
Put my name on it!

Also, this may not be the correct place to ask this, but is there a REP or anything in place regarding rosparam and dynamic reconfigure parameter naming conventions? Im not sure if this would be useful for anyone else, but having a feature where you can dump 'tuned' variables using dynamic reconfigure into a yaml, that you then pass to say a launch file which loads the dumped 'tuned' parameters load into the rosparam server as many nodes initialize themselves by grabbing rosparams. This cant really be done as many packages who do support dynamic reconfigure AND initialize by rosparam server parameters since the naming of variables between the two features is not consistent (for example move_base)

renrut, 2016-03-01:
@l0g1x you ever figure it out?

vagvaz, 2016-05-27:
HI I created a pull request it is a simple is not --> != change in the client

mikaelarguedas, 2016-06-12:
fixed in #61

closed 2016-06-12 by mikaelarguedas

lindemeier, 2016-07-27:
I am experiencing this issue on Ubuntu 16.04 (kinetic).

```
!!python/object/new:dynamic_reconfigure.encoding.Config
dictitems:
  autoScale: true
  debayerLibraw: true
  groups: !!python/object/new:dynamic_reconfigure.encoding.Config
    dictitems:
      autoScale: true
      debayerLibraw: true
      groups: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      id: 0
      name: Default
      parameters: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      parent: 0
      rawBits: 16
      state: true
      streamingMode: false
      type: ''
      whiteBalanceCoeff_b: 1.0
      whiteBalanceCoeff_g0: 1.0
      whiteBalanceCoeff_g1: 1.0
      whiteBalanceCoeff_r: 1.0
    state: []
  rawBits: 16
  streamingMode: false
  whiteBalanceCoeff_b: 1.0
  whiteBalanceCoeff_g0: 1.0
  whiteBalanceCoeff_g1: 1.0
  whiteBalanceCoeff_r: 1.0
state: []
```

Loading the parameters produces this error:
```
error updating parameters: don't know parameter: groups
```

Removing the duplicates solves it
```
!!python/object/new:dynamic_reconfigure.encoding.Config
dictitems:
  autoScale: false
  debayerLibraw: true  
  rawBits: 16
  streamingMode: false
  whiteBalanceCoeff_b: 1.0
  whiteBalanceCoeff_g0: 0.5
  whiteBalanceCoeff_g1: 0.5
  whiteBalanceCoeff_r: 1.0
state: []
```

jacquelinekay, 2016-11-11:
This fix persists in the debian because dynamic_reconfigure needs to be released for Kinetic.

mikaelarguedas, 2016-11-15:
The upcoming sync should propagate the fix to the main apt repo for all ros distributions

Kaju-Bubanja, 2019-08-06:
This issue still exists in melodic, any release schedule for this issue?

130s, 2020-01-14:
My team has been troubled with this issue (and potentially other issues regarding `dump` feature) for a few years and internally have come up with a workaround, which has been causing a great amount of technical debt and new issues.

#61 seems to have become available since [1.5.44 on Kinetic](http://docs.ros.org/kinetic/changelogs/dynamic_reconfigure/changelog.html) back in 2016 (I'm on ROS K) but our problem has persisted.

I'm willing to provide more info but I'd request reopen this ticket (or I can open a separate one).