ros/dynamic_reconfigure #57: Test fails on Arch Linux

bchretien, 2016-03-10:
While compiling `dynamic_reconfigure` on Arch Linux (e.g. with indigo), we get:

```txt
[  5%] Generating dynamic reconfigure files from cfg/Test.cfg: /home/user/dev/dynamic_reconfigure/build/devel/include/dynamic_reconfigure/TestConfig.h /home/user/dev/dynamic_reconfigure/build/devel/lib
/python2.7/site-packages/dynamic_reconfigure/cfg/TestConfig.py
[  5%] Built target _dynamic_reconfigure_generate_messages_check_deps_ParamDescription
Traceback (most recent call last):
  File "/home/user/dev/dynamic_reconfigure/cfg/Test.cfg", line 37, in <module>
    from dynamic_reconfigure.parameter_generator_catkin import *
ImportError: No module named 'dynamic_reconfigure'
CMakeFiles/dynamic_reconfigure_gencfg.dir/build.make:63: recipe for target 'devel/include/dynamic_reconfigure/TestConfig.h' failed
make[2]: *** [devel/include/dynamic_reconfigure/TestConfig.h] Error 1
make[2]: Target 'CMakeFiles/dynamic_reconfigure_gencfg.dir/build' not remade because of errors.
CMakeFiles/Makefile2:584: recipe for target 'CMakeFiles/dynamic_reconfigure_gencfg.dir/all' failed
```

We currently add `-DCATKIN_ENABLE_TESTING=OFF` to avoid the issue. See the [original discussion](https://aur.archlinux.org/packages/ros-indigo-dynamic-reconfigure/) in the AUR.

mikaelarguedas, 2017-06-08:
The problem is IMO that the way we build and test of the buildfarm does avoid this problem (we build and install the package and then build the tests and run them).

The possible solutions are (from the ugliest to the nicest):
- modify the test to append the right location to sys.path so that it can include itself
- use -DCATKIN_ENABLE_TESTING=OFF when building and activate it only for testing
- move the tests into a designated test package to run all "module level test". This way tests don't have to be released and the test package will depend on dynamic_reconfigure so it will already be part of the python path when the test package is built and ran

@allenh1 FYI follow up of #85

allenh1, 2017-06-08:
@mikaelarguedas Thank you very much. I can confirm the fix is to disable the flag.

allenh1, 2017-07-17:
https://github.com/ros/ros-overlay/blob/master/eclass/ros-cmake.eclass#L158-L166

This was the fix for me. There's some real inconsisency: on Ubuntu, python libraries all go to the lib folder, but on Arch and Gentoo this is not so. Those lines essentially force the files to map into lib64.

mikaelarguedas, 2017-07-17:
@bchretien  I'm going to close this. I cross posted @allenh1 workaround on the original AUR post as well. Feel free to comment here if this is still a problem and we can re-open

mikaelarguedas, 2017-07-17:
@allenh1 FYI: I edited the link in https://github.com/ros/dynamic_reconfigure/issues/57#issuecomment-315894131 to point to a specific commit rather than master for this answer to be more future-proof

closed 2017-07-17 by mikaelarguedas