ros/dynamic_reconfigure #62: Dynamic dynamic reconfigure for python

awesomebytes, 2016-05-31:
I implemented a tiny layer to be able to do a "dynamic dynamic reconfigure" server at https://github.com/pal-robotics/ddynamic_reconfigure_python 

It allows to `add` some dynamic reconfigure variables and then just call a `start` method with a callback to have a dynamic reconfigure server running (instead of using a `.cfg` file and all the stuff necessary currently for having the server). Give a look to the README with an example.
Maybe we want this to be added to the dynamic_reconfigure package itself?

If you give a look at the code I needed to write to achieve this, it's very short: https://github.com/pal-robotics/ddynamic_reconfigure_python/blob/master/src/ddynamic_reconfigure_python/ddynamic_reconfigure.py

@ibaranov-cp I think you already tried it and liked it too :)

ibaranov-cp, 2016-05-31:
I tired it, quite liked it actually. From a demo/quick prototype phase, very nice to be able to just write pure Python, and still use dynamic reconfigure for tuning.

mikaelarguedas, 2016-06-13:
@awesomebytes Nice work, that's a cool feature addition. In order to integrate it to the dynamic_reconfigure package we would need to provide the same feature for c++ to keep the feature set symmetrical. If nobody has time to implement the c++ counterpart, it can be released as a standalone package (like `ddynamic-reconfigure-python`) depending on dynamic_reconfigure.

awesomebytes, 2016-06-13:
Internally at PAL we have a c++ version. I'll check if it can be released
somehow :)

mikaelarguedas, 2016-06-13:
Great! thanks

NikolausDemmel, 2016-06-29:
> Internally at PAL we have a c++ version. I'll check if it can be released somehow :)

I was just looking for something like this and came across this issue. My use-case is exposing the configuration of an existing application to ROS (as a developer interface). The configuration is currently specified programmatically in the source code itself such that all the type information is available at runtime and could be used to feed dynamic reconfigure. Your ddynamic-reconfigure looks like exactly the tool I would need for that, so if releasing it would be a possibilty, that would be pretty neat!

NikolausDemmel, 2016-08-15:
> Internally at PAL we have a c++ version. I'll check if it can be released somehow :)

@awesomebytes, any updates on the above?

awesomebytes, 2016-08-16:
I'm on holidays, I don't expect to get near a computer soon, sorry!

NikolausDemmel, 2016-08-16:
Sure, no worries, enjoy!

NikolausDemmel, 2016-10-08:
@awesomebytes, could you somehow get me access to that C++ version?

awesomebytes, 2016-10-08:
@NikolausDemmel & @mikaelarguedas as spoken personally in ROSCON, I've given you access to a bitbucket repo of the current implementation so we can move it from supporting int's and double's to support all of dynamic reconfigure types (bools, strings, evil_enumerates). Let's see if we can make it work soon and add everything directly to dynamic_reconfigure.

NikolausDemmel, 2016-10-10:
@awesomebytes, cheers, I started work on this. Since I need a bit more than just the basic "here is a pointer, please update the variable directly when the reconfigure service is called", I'm considering to take ideas from the currently implementation but effectively write it from scratch. If course I would like to retain the simple use cases in the same style as they are now.

Where would be a good place to discuss design and implementation (in case someone is interested)?

cc @mikaelarguedas

ubuntuslave, 2017-01-04:
@NikolausDemmel, @awesomebytes, How is it going? Any plans on releasing the C++ version of [ddynamic_reconfigure](https://github.com/pal-robotics/ddynamic_reconfigure_python)?

awesomebytes, 2017-01-04:
Hello Carlos,

I haven't looked into it again, it got some interest at the moment but
nothing was finished. Maybe your comment will ping my friends? :)

mikaelarguedas, 2017-01-10:
I didn't have any time to work on the C++ implementation since we mentionned it at ROSCon. @NikolausDemmel started working on it a while back but I don't know what the status is

NikolausDemmel, 2017-01-18:
Unfortunatly I didn't ever find the time to continue work here. Concepts I have, and use cases, just no time to implement :-|.

linknum23, 2017-11-24:
As mainly a c++ dev I have been interested in adding this for a bit now. @NikolausDemmel it seems like you have some ideas on the implementation and use cases. I am willing to start from scratch on this but if theres something available to start with it might come together faster...

NikolausDemmel, 2017-11-27:
@linknum23, I have nothing to base any further work on. Maybe @awesomebytes can give you access to his internal version from a few years ago.

awesomebytes, 2017-11-27:
@linknum23 @NikolausDemmel let me give it a try asking.

awesomebytes, 2017-11-27:
Ok, I didn't really remember how was the status of this. I added you to a private repository with the partial implementation. The caveats of it:

> the current implementation so we can move it from supporting int's and double's to support all of dynamic reconfigure types (bools, strings, evil_enumerates)

There is a branch by @NikolausDemmel called niko-devel with some changes.

linknum23, 2017-11-27:
Fantastic. Thanks @awesomebytes! I will start diving into this.

NikolausDemmel, 2017-11-27:
Ah, see, I didn't even remember I pushed this :-). Thanks @awesomebytes and have fun @linknum23.

awesomebytes, 2017-11-29:
@linknum23 I ignore it it will be useful but an implementation of the dynamic reconfigure client for c++ was merged relatively recently: https://github.com/ros/dynamic_reconfigure/pull/78

linknum23, 2017-11-29:
Thanks, this has the potential to be useful for dd_reconfigure. The c++ client looks well documented.

awesomebytes, 2018-02-27:
Hey @NikolausDemmel @linknum23 did you get any work done on this? :)

NikolausDemmel, 2018-02-27:
I haven't worked on it... :-/

wew84, 2018-05-29:
My team and I are encountering these limits to dynamic reconfigure as well. We would would like to build a full python/cpp fix to it. @awesomebytes it would be great to have something to start with. Any chance of getting access to your Bitbucket as well?

awesomebytes, 2018-05-29:
@wew84 hello! I can give you access, could you drop me a line of who you are and what you do (maybe by email if you prefer) just to have a reference of who has access to it. I don't own the specific intellectual property of it (even tho I'm allowed to share it privately for the purpose of creating this open source tool) so I need to at least know who you are. Thank you :)

awesomebytes, 2018-06-06:
> Thank you @awesomebytes <https://github.com/awesomebytes> for giving me
> access to the repo. We are currently working on a fairly large overhaul of
> dynamic_reconfigure where we will be implementing the following:
>
>    1. C++ API for dynamic_reconfigure client.
>    2. dDynamic_reconfigure in all of its functionality for both python
>    and C++
>    3. 3d_reconfigure: a new feature where the clients will be able to
>    change the limits, defaults, and whether a param is enabled or not
>
> We are trying to do this over the course of the next month or two. And
> would be happy to hear your input on this

I believe there is a C++ client API already, I'm on mobile so I can't link
it... But it should be in this repo!

And there is a Python dynamic dynamic reconfigure server too (which I
wrote), are you going to rewrite it? Why?

3d configure sounds cool! (3d as dynamic dynamic dynamic? Haha)

wew84, 2018-06-06:
Looked more at the source of the repo and see the C++ client API. It doesn't seem to be documented (at least not that I could find), will look into it...

I more meant to write the C++ side and merge the two repos and make sure that they work seamlessly together. No reason to reinvent the wheel.

awesomebytes, 2018-06-06:
Sounds great Noam!

Count with me if you need any help with the Python part or we need to
discuss anything to keep the same API. Currently the Python package is a
separate package, so we can expose a similar API in both languages if
needed.

About the C++ client API, documentation, maybe it's time to improve it! Haha

flixr, 2018-09-21:
I'm also quite interested in this... any updates here?

Didn't even know the pal-robotics dynamic dynamic_reconfigure, but I basically also implemented something similar in python.
Now we would want to use this in some of our C++ nodes as well, so before I start implementing the same again...

wew84, 2018-09-21:
Our C++ side is finished, and created a pull request to @awesomebytes repo. I haven't had time to discuss further changing the Python API, but would be happy to discuss it. 

The C++ side of 3d_reconfigure is also done and can be found [here](https://github.com/wew84/3d_reconfigure).

flixr, 2018-09-30:
@wew84 sorry, but which repo exactly do you mean?
Seems it is not https://github.com/pal-robotics/ddynamic_reconfigure

awesomebytes, 2018-10-02:
@flixr I wasn't aware PAL Robotics released a version, I had a private copy I had permission to share with interested people to develop a fully working solution. And that's what I did with @wew84 @NikolausDemmel @linknum23 

Now I presume I can open publicly the repo (please confirm @v-lopez even tho I contacted you thru other means) as they share the same commit history up until the latest version I had access.

Here you can find the version from @wew84 : https://github.com/awesomebytes/ddynamic_reconfigure based on the 0.0.5 version from PAL. Now they are up to 0.1.4.

I think there should be a quick discussion in between @v-lopez and @wew84 about the capabilities of both (given the one in the [ddynamic_reconfigure in the PAL repo](https://github.com/pal-robotics/ddynamic_reconfigure) has no README right now).

v-lopez, 2018-10-02:
Yes @awesomebytes no problems in publishing it now.

I've glanced over @wew84, and my impression is that its purpose and use differs significantly. 

Our view of ddynamic_reconfigure was to make it less verbose to use (probably some documentation about it on our side  would've made it more evident).

So for instance:
```cpp
  int int_variable = 0;
  ros::NodeHandle nh("~");
  ddynamic_reconfigure::DDynamicReconfigure dd(nh);
  dd.RegisterVariable(&int_variable, "My int variable", -10000, 10000);
  dd.PublishServicesTopics();
```

These 5 lines will make the int_variable modifiable from the standard dynamic_reconfigure interfaces.
Considering that the first line is the variable declaration and the DDynamicReconfigure object needs only to be created once,  the last line needs to be only called once after registering everything, you can register N variables with N+2 new lines of code.

Since we provide the address of the variable, the variable is modified in-place automatically. You still can control when this happens if you control the queue associated to the nodehandle provided to the DDynamicReconfigure.

Our documentation is completely lackluster and that is something we'll work on, but the code is reliable and used extensively in all our robots.

v-lopez, 2018-10-02:
Thanks for clarifying, but from my understanding, you still need to define a callback and in the code of that callback associate write it to the variable.

It may not seem much but it makes it so much more comfortable to use. We can try to merge your changes into ours, but at the moment we cannot afford to break our API.

Noam-Dori, 2018-10-02:
@v-lopez there is no need to declare a callback. 
Instead of doing ``dd.start(callback);``,
you can write ``dd.start();``

I don't believe any of the changes will break the API which cannot be fixed via refactoring & renaming.

If there are any other issues be sure to let me know

v-lopez, 2018-10-02:
If there is no callback how do you receive the modified values? Do you need to query the dd object?

v-lopez, 2018-10-02:
Feel free to open a merge request and we can discuss this there, and see how we can make it work.

Noam-Dori, 2018-10-02:
First, the modified values are automatically changed whether or not a callback exists. This behavior is defined both in the original dynamic_reconfigure (for C++ at least) and in the original code of ddynamic_reconfigure. The callback is an extra.

From there, there are two additional ways to fetch the modified values: directly from the DDynamicReconfigure instance, or via a callback.

if you want the variables right when they change, you use the callback, for which you need to define a function with the signature:
```cpp
void callback(const DDMap& map, int level)
```
and within said callback, use either of the following utility methods:
```cpp
// option 1 - get: retrieves a physical value of a parameter by name
    int number = ddynamic_reconfigure::get(map,"int_param").toInt();
// option 2 - at: retrieves a parameter pointer so that you can modify the parameter's behavior (or get its value)
    int number_again = ddynamic_reconfigure::at(map,"int_param")->getValue().toInt();
```
This way is very similar to the way dynamic_reconfigure defines its callbacks.

In a similar manner, you can get the parameters/values directly from the DDynamicReconfigure instance:
```cpp
// assume we have fully defined DDynamicReconfigure instance dd
// option 1 - get: retrieves a physical value of a parameter by name
int number = dd.get(map,"int_param").toInt();
// option 2 - at: retrieves a parameter pointer so that you can modify the parameter's behavior (or get its value)
int number_again =dd.at(map,"int_param")->getValue().toInt();
```
With any of the strategies listed above you can do what you wish with them.

Noam-Dori, 2018-10-04:
Ok, I now understand the issue at hand.

Luckily, the generic parameter class may have multiple implementations and extensions, some of which can support both the legacy implementation of parameter registration and the current one:
To simulate the pointer strategy we can extend any instance of ``DDParam`` to a sort of ``DDPointerParam``. For example, here I implemented two ways to do so for integers:
**option 1 - ``DDIntRef``:**
```cpp
#include <ddynamic_reconfigure/param/dd_int_param.h>
#include <ddynamic_reconfigure/dd_param.h>

namespace my_dd_reconfig {
    // interface definition
    class DDRef : virtual public DDParam {
        virtual void *getReference();
    };
    // class definition
    class DDIntRef : public DDRef, public DDInt {
    public:

        int *getReference();

        /**
         * creates a new int reference param
         * @param description details about the parameter
         * @param max the maximum allowed value. Defaults to INT32_MAX
         * @param min the minimum allowed value. Defaults to INT32_MIN
         */
        inline DDIntRef(const string &description, int min = INT32_MIN, int max = INT32_MAX) :
                DDInt(description,0,description,0,min,max) {};
    };

    int *DDIntRef::getReference() {
        return &val_;
    };
}
```
this strategy makes sure that the DDynamic instance is the one tasked with the memory management of the parameters and uses a bit less memory than option 2.
Example of how to use it:
```cpp
ros::NodeHandle nh;
ddynamic_reconfigure::DDynamicReconfigure dd(nh);
DDIntRef int_param("My int variable", -10000, 10000);
int *int_variable = int_param.getReference(); // pointer memory managed by int_param
dd.add(int_param); // at this point int_param is now managed by the DDynamic instance
dd.start();
```
the disadvantege of this is that this strategy does not fully match your old API (as the variable is now a pointer and not a reference)

**option 2 - ``DDIntPointer``:**
```cpp
#include <ddynamic_reconfigure/param/dd_int_param.h>

namespace my_dd_reconfig {
    class DDPointer : virtual public DDParam {};
    // class definition
    class DDIntPointer : public DDPointer, public DDInt {
    public:

        void setValue(Value val);

        /**
         * creates a new int pointer param
         * @param pointer a pointer to an integer to update.
         * @param description details about the parameter
         * @param max the maximum allowed value. Defaults to INT32_MAX
         * @param min the minimum allowed value. Defaults to INT32_MIN
         */
        inline DDIntPointer(int *pointer, const string &description, int min = INT32_MIN, int max = INT32_MAX) :
                DDInt(description,0,description,0,min,max) {
            other_pointer_ = pointer;
        };

    protected:
        int *other_pointer_;
    };

    void DDIntPointer::setValue(Value val) {
        val_ = val.toInt();
        if(other_pointer_ != NULL) { // prevents some unwanted segfaults
            *other_pointer_ = val_;
        }
    };
}
```
this strategy matches exactly what existed before in the old API, and can be modified a bit to store additional pointers. The pointer you provide is managed by you so you have the power over it.
Example of how to use it:
```cpp
int int_variable = 0;
ros::NodeHandle nh;
ddynamic_reconfigure::DDynamicReconfigure dd(nh);
dd.add(new DDIntPointer(&int_variable, "My int variable", -10000, 10000));
dd.start();
```
the disadvantage is that there is additional memory to create per instance, you need to manage pointer memory.

doronhi, 2018-10-15:
Hi @wew84,
Is a python version also available? I couldn't find it.

awesomebytes, 2018-10-15:
@doronhi The main repo  is at https://github.com/pal-robotics/ddynamic_reconfigure_python

Hmm it's weird you got an empty package. I just tried in a fresh machine to do your command and it installed correctly. I could run `rosrun ddynamic_reconfigure_python example.py` successfully

doronhi, 2018-10-15:
I did catkin_make and now it works.
Do you know if it also works with ROS2?

closed 2025-05-31 by sloretz