ros/dynamic_reconfigure #69: Shebang in setup_custom_pythonpath.sh.in

follesoe, 2016-12-08:
I've run into some issues while building an arm Docker image to run my ROS application. I think the issue has to do with missing shebang (`#!/usr/bin/env bash`) in the `cmake/setup_custom_pythonpath.sh.in` file.

I think it's related to how Docker executes RUN statements - see http://kimh.github.io/blog/en/docker/gotchas-in-writing-dockerfile-en/ and paragraph on exec format error.

I only run into the issue in my arm images, as they are build using the technique described here: https://resin.io/blog/building-arm-containers-on-any-x86-machine-even-dockerhub/ (using qemu).

My project has been building fine until I introduced dynamic_reconfigure - and after trying to isolate the problem, I have been able to build in my docker container if I add the shebang to the generated `setup_custom_pythonpath.sh` file.

Forgive me if my understanding of the problem is incorrect, but would it make sense to include the shebang in the template file used to generate the `setup_custom_pythonpath.sh` file?

mikaelarguedas, 2017-03-14:
@follesoe sorry for the late reply.
It seems to work fine on the [ros buildfarm](http://build.ros.org/view/Kbin_uxv8_uXv8/job/Kbin_uxv8_uXv8__dynamic_reconfigure__ubuntu_xenial_arm64__binary/8/) and locally, could you provide a Dockerfile to reproduce this ? Thanks!

joshughes, 2017-07-11:
Can confirm this issue with qemu and cross compiling

joshughes, 2017-07-11:
```
FROM armv7/armhf-ubuntu:xenial

ENTRYPOINT [ "/usr/bin/qemu-arm-static", "/usr/bin/env", "QEMU_EXECVE=1" ]
CMD [ "/bin/bash" ]
ENV QEMU_EXECVE 1

COPY qemu/resin-xbuild qemu/qemu-arm-static  /usr/bin/

RUN [ "qemu-arm-static", "/bin/sh", "-c", "ln -s resin-xbuild /usr/bin/cross-build-start; ln -s resin-xbuild /usr/bin/cross-build-end; mv /bin/sh /bin/sh.real; ln -s /bin/sh.real /bin/sh" ]

RUN apt-get update && apt-get install -y lsb-release &&\
    sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list' &&\
    apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116 &&\
    apt-get update -y &&\
    apt-get upgrade -y

RUN [ "cross-build-start" ]

# Install development packages
RUN apt-get install \
    build-essential \
    debhelper \
    libhttp-parser-dev \
    liblz4-dev \
    libnatpmp-dev \
    dh-systemd \
    ruby-ronn \
    ccache \
    g++ \
    make \
    devscripts \
    libsdl-image1.2-dev \
    python-gevent \
    libunwind8-dev \
    expect-dev \
    python-wstool \
    python-rosdep \
    ninja-build \
    ros-kinetic-ros-base \
    ros-kinetic-urdf \
    ros-kinetic-xacro \
    ros-kinetic-navigation \
    ros-kinetic-rgbd-launch \
    ros-kinetic-filters \
    ros-kinetic-depthimage-to-laserscan \
    ros-kinetic-tf2-sensor-msgs \
    ros-kinetic-diagnostic-updater \
    ros-kinetic-camera-info-manager \
    ros-kinetic-rviz \
    netpbm \
    libusb-1.0-0-dev \
    -y

ENV CCACHE_DIR /.ccache

RUN [ "cross-build-end" ]

```

joshughes, 2017-07-11:
The key here is to be running a command through qemu... which wraps the environment and fails at this bash script

mikaelarguedas, 2017-07-11:
Thanks @joshughes for following up. I created #91 to address this. Can you confirm that it solves the issue for you ?

mikaelarguedas, 2017-07-18:
fixed in #91 thanks @joshughes for testing

closed 2017-07-18 by mikaelarguedas