ros/dynamic_reconfigure #70: Feature Request: Reconfigure the white balance the same way it is possible in the wxPropView GUI

Kaju-Bubanja, 2016-12-08:


mikaelarguedas, 2017-01-03:
Hi @Kaju-Bubanja, 
I'm not sure I understand your request exactly. Could you give more context please?

As I understand it, it seems that you want to be able to reconfigure the white balance of a camera, is that right? If that's the case you'll need to expose the white balance parameter in your camera driver or whatever node is publishing the images.
Parameters tunable with `dynamic_reconfigure` need to be defined in a `.cfg` file in your camera node, some examples and tutorials are avialable [here](http://wiki.ros.org/dynamic_reconfigure/Tutorials)

Hope this helps

mikaelarguedas, 2017-03-08:
without further detail and because the issue doesn't seem to be related to `dynamic_reconfigure` itself I'm going to close this. @Kaju-Bubanja Feel free to comment here and reopen this if you have more details on this issue.

closed 2017-03-08 by mikaelarguedas