ros/dynamic_reconfigure #72: Allow dynamic_reconfigure to fail/reject a change

CodeFinder2, 2017-01-16:
Hi all,

I want to suggest a minor improvment: currently, the callback's return type which updates the parameter is `void`. Thus, **if the provided (new) parameters are invalid and nothing will change, there is no possibility to indicate this**. I would therefore **suggest to change the return type of the callback to `bool`** and pass that information to the underlying `set_parameters` service, see https://github.com/ros/dynamic_reconfigure/blob/master/include/dynamic_reconfigure/server.h#L223 (and L227). If changing parameters in the callback fails, the service caller would get to know that as well.

As a **quick fix (because I need this feature urgently)**, would it be possible to add a boolean parameter to the `.cfg` file so that that callback modifiers this boolean to indicate success/failure? What I've understood from  the code (see https://github.com/ros/dynamic_reconfigure/blob/master/include/dynamic_reconfigure/server.h#L226 ), the `new_config` (passed to the callback) is stored in the service response. Do you think that this workaround would do the trick? Or is there even a simpler solution that I overlooked completely?

What do you think about that? Glad to hear your comments.

progtologist, 2017-03-14:
You can reject the reconfiguration:
```cpp
void reconfigureCallback(config &config, uint32_t level) {
  if (accepted) {
    config_ = config;
  } else {
    config = config_;
  }
}
```
The reconfiguration parameters are passed by reference, any changes you apply to the them are reflected on the clients as well.

CodeFinder2, 2017-03-14:
Thanks and sorry for the noise!

closed 2017-03-14 by CodeFinder2