ros/dynamic_reconfigure #75: New check of description is causing other ros package to fail

tspicer01, 2017-03-13:
We are seeing errors running with the latest version of dynamic_reconfigure as the new check is causing failures.   The latest version of ros image_transport_plugins (1.9.5) fails with the following error:

 ```
File "/tmp/buildd/workspace/dynamic_reconfigure/parameter_generator_catkin.py", line 70, in check_description
raise Exception(r"""quotes not allowed in description string `%s`""" % description)
Exception: quotes not allowed in description string `Try to achieve either 'target_bitrate' or 'quality'`
make[4]: *** [/tmp/buildd/workspace/theora_image_transport/include/theora_image_transport/TheoraPublisherConfig.h] Error 1
```

Is there a way to allow for backward compatibility or before turning the check on ensure that all other packages are compliant?

Any help with this would be appreciated.

mikaelarguedas, 2017-03-13:
Thanks @tspicer01 for reporting it and the pointer. The check may be too restrictive. I'll perform some testing and get back to you.

mikaelarguedas, 2017-03-13:
@tspicer01 @mikepurvis Sorry for the trouble, #76 should fix it, I tried it on desktop_full and it built fine. Can you try it on your side and comment here with your results ? Thanks!

tspicer01, 2017-03-13:
Thanks for the quick response and turnaround.

I've tried it on our end and using your branch from #76 the packages that were failing are now successful.

mikaelarguedas, 2017-03-13:
Great, thanks for testing it!

PR #76 (2017-03-13): remove checks for quotes in non const or enum descriptions
Relax constraint on descriptions added in #74 , fixes #75
closed 2017-03-13 by mikaelarguedas