ros/dynamic_reconfigure #79: -Wshadow warnings on generated dynamic reconfigure options files.

jeffeb3, 2017-03-16:
I'm on Ubuntu 16.04, ROS kinetic, but this issue also affects my 14.04/indigo machine.

I'm generating config options using `generate_dynamic_reconfigure_options` and it's making a header file with shadow variables, which is important because my nightly build machine uses `-Wshadow and -Werror` flags to build my code.

```
warning: declaration of ‘edit_method’ shadows a member of ‘code_package::adaptive_estimationConfig::ParamDescription<T>’ [-Wshadow]

warning: declaration of ‘description’ shadows a member of ‘code_package::adaptive_estimationConfig::ParamDescription<T>’ [-Wshadow]
...
```
Also errors for level, type, name, id, parent

Should be an easy fix.

mikaelarguedas, 2017-03-16:
Thanks @jeffeb3 for reporting and fixing this.

mikaelarguedas, 2017-03-16:
fixed in #80

closed 2017-03-16 by mikaelarguedas

jeffeb3, 2017-03-16:
Your welcome. Thank you for merging it so fast. I have a dumb question. How long does it take for this to end up in the apt packages on my machine? If it's more than about a week, I should figure out how to install this from source.

mikaelarguedas, 2017-03-16:
I'm planning on doing a new release of this package in the next few days/week, if you are using the main ros repository it will likely take more than a week before being available through apt.
If you want to use the lastest packages released, there is a "testing" repository called [shadow-fixed](http://wiki.ros.org/ShadowRepository) where all packages are synchronized as soon as they are released  (in that case it should meet your timeframe).

As you mentionned, you can always build it from source. You can [create a catkin workspace](http://wiki.ros.org/catkin/Tutorials/create_a_workspace) and clone this repository in the src folder. Then running `catkin_make` will do the job. Just remind to source your workspace before using it (by running `source devel/setup.bash`)