ros/dynamic_reconfigure #81: getCurrentConfiguration() sometimes fails in new cpp client

JVSelvag, 2017-03-23:
In client.h: getCurrentConfiguration() sometimes fails to get the current configuration from the dynamic reconfigure server if the server has been recently updated with a new configuration.

A call to getCurrentConfiguration() will return when the "received_configuration_" boolean flag is true. 
However, this flag is only set to false in the constructor, so getCurrentConfiguration() will only work correctly the first time. Thereafter it will always return immediately, and not wait for the callback from the server update.

Possible workaround: Set "received_configuration_" to false in the method setConfiguration().

JVSelvag, 2017-03-23:
Also in client.h: getDescription() suffers from the same problem. Here, the boolean flag "received_description_" is only set to false in the constructor of the Client() class.

JVSelvag, 2017-03-23:
However, since the description won't change, getDescription() might work as it is...

mikaelarguedas, 2017-03-23:
Thanks @JVSelvag for reporting this. #82 should fix it

PR #82 (2017-03-23): reset received_configuration_ for every request sent
fixes #81
closed 2017-03-23 by mikaelarguedas