ros/dynamic_reconfigure #85: Compiler Error on Gentoo

allenh1, 2017-06-08:
Remember when I said that `dynamic_reconfigure` emerged on Gentoo? Well, it did, but... Not so much anymore (the new ebuild format is pure CMake and doesn't call catkin).

```
Scanning dependencies of target dynamic_reconfigure_gencfg
make[2]: Leaving directory '/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build'
make -f CMakeFiles/dynamic_reconfigure_gencfg.dir/build.make CMakeFiles/dynamic_reconfigure_gencfg.dir/build
[  0%] Built target _dynamic_reconfigure_generate_messages_check_deps_SensorLevels
make[2]: Entering directory '/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build'
make[2]: Nothing to be done for 'CMakeFiles/std_msgs_generate_messages_nodejs.dir/build'.
make[2]: Leaving directory '/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build'
make[2]: Entering directory '/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build'
Dependee "/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build/CMakeFiles/dynamic_reconfigure_config_init_mutex.dir/DependInfo.cmake" is newer than depender "/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build/CMakeFiles/dynamic_reconfigure_config_init_mutex.dir/depend.internal".
Dependee "/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build/CMakeFiles/dynamic_reconfigure_config_init_mutex.dir/depend.internal".
Scanning dependencies of target dynamic_reconfigure_config_init_mutex
make[2]: Leaving directory '/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build'
make -f CMakeFiles/dynamic_reconfigure_config_init_mutex.dir/build.make CMakeFiles/dynamic_reconfigure_config_init_mutex.dir/build
[  0%] Built target std_msgs_generate_messages_nodejs
[  1%] Generating dynamic reconfigure files from cfg/Test.cfg: /var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build/devel/include/dynamic_reconfigure/TestConfig.h /var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build/devel/lib64/python3.5/site-packages/dynamic_reconfigure/cfg/TestConfig.py
make[2]: Entering directory '/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build'
catkin_generated/env_cached.sh /var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build/setup_custom_pythonpath.sh /var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48/cfg/Test.cfg /var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48 /var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build/devel/share/dynamic_reconfigure /var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build/devel/include/dynamic_reconfigure /var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build/devel/lib64/python3.5/site-packages/dynamic_reconfigure
[  3%] Building CXX object CMakeFiles/dynamic_reconfigure_config_init_mutex.dir/src/dynamic_reconfigure_config_init_mutex.cpp.o
/usr/bin/x86_64-pc-linux-gnu-g++  -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"dynamic_reconfigure\" -Ddynamic_reconfigure_config_init_mutex_EXPORTS -I/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build/devel/include -I/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48/include -I/opt/ros/lunar/include -I/opt/ros/lunar/share/xmlrpcpp/cmake/../../../include/xmlrpcpp   -march=core-avx2 -O3 -pipe -std=c++11 -fPIC   -o CMakeFiles/dynamic_reconfigure_config_init_mutex.dir/src/dynamic_reconfigure_config_init_mutex.cpp.o -c /var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48/src/dynamic_reconfigure_config_init_mutex.cpp
Traceback (most recent call last):
  File "/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48/cfg/Test.cfg", line 37, in <module>
    from dynamic_reconfigure.parameter_generator_catkin import *
ImportError: No module named 'dynamic_reconfigure'
make[2]: Leaving directory '/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build'
make[2]: *** [CMakeFiles/dynamic_reconfigure_gencfg.dir/build.make:64: devel/include/dynamic_reconfigure/TestConfig.h] Error 1
make[2]: Leaving directory '/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build'
make[1]: *** [CMakeFiles/Makefile2:1477: CMakeFiles/dynamic_reconfigure_gencfg.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  3%] Built target _dynamic_reconfigure_generate_messages_check_deps_Reconfigure
make[2]: Leaving directory '/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build'
[  3%] Built target _dynamic_reconfigure_generate_messages_check_deps_Config
[  4%] Linking CXX shared library devel/lib/libdynamic_reconfigure_config_init_mutex.so
/usr/bin/cmake -E cmake_link_script CMakeFiles/dynamic_reconfigure_config_init_mutex.dir/link.txt --verbose=1
/usr/bin/x86_64-pc-linux-gnu-g++  -fPIC -march=core-avx2 -O3 -pipe -std=c++11 -Wl,-O1 -Wl,--as-needed -shared -Wl,-soname,libdynamic_reconfigure_config_init_mutex.so -o devel/lib/libdynamic_reconfigure_config_init_mutex.so CMakeFiles/dynamic_reconfigure_config_init_mutex.dir/src/dynamic_reconfigure_config_init_mutex.cpp.o -lboost_system-mt -lboost_thread-mt -lboost_chrono-mt -lboost_date_time-mt -lboost_atomic-mt 
make[2]: Leaving directory '/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build'
[  4%] Built target dynamic_reconfigure_config_init_mutex
make[1]: Leaving directory '/var/tmp/portage/ros-lunar/dynamic_reconfigure-1.5.48/work/dynamic_reconfigure-1.5.48_build'
make: *** [Makefile:128: all] Error 2
```

It goes without saying that this is should only be seen as low priority, but, should you have any ideas, please let me know!

allenh1, 2017-06-08:
What's odd to me is that the module is tyring to import itself... could it be that it's supposed to be installed at this point, and it's not in the python path?

mikaelarguedas, 2017-06-08:
duplicate of #57

closed 2017-06-08 by mikaelarguedas

mikaelarguedas, 2017-06-08:
I'm going to leave this closed and comment on the other issue. Thanks for reporting!