ros/dynamic_reconfigure #86: Allow using dyn. reconfig. with -Werror=float-equal

CodeFinder2, 2017-06-10:
Hi all,

I am using dynamic_reconfigure with much stricter compiler settings, namely `-Werror=float-equal` (among others). Unfortunately, this results in errors like:
```
.../MyPackageConfig.h: In instantiation of ‘void my_namespace::MyPackageConfig::ParamDescription<T>::calcLevel(uint32_t&, const my_namespace::MyPackageConfig&, const my_namespace::MyPackageConfig&) const [with T = double; uint32_t = unsigned int]’:
.../my_package.cpp:371:1:   required from here
.../MyPackageConfig.h:77:28: error: comparing floating point with == or != is unsafe [-Werror=float-equal]
         if (config1.*field != config2.*field)
                            ^
cc1plus: all warnings being treated as errors
```

One can get rid of these problems in 2 ways:
 1. move all *.cfg files to separate packages and
 2.  "patch" the include directory to generated headers in CMake.

Both solutions are somewhat "hacks". (a) requires to create a new ROS package just for the .cfg file (overhead, file bloat). (b) is hackish in a sense that additional CMake code is required which may also hide other issues.

Therefore I would suggest patching the [template](https://github.com/ros/dynamic_reconfigure/blob/master/templates/ConfigType.h.template#L77) to allow such flags. (My first idea would be a specialized version of the template for floats and doubles but there's probably also a smarter / shorter way.)

Any comments on this?

eric-wieser, 2018-01-10:
It seems like cmake should be using `SYSTEM` in its includes, which cause warnings to be ignored in the specified files

eric-wieser, 2018-01-10:
I think I might be, yes.

ernestmc, 2024-02-01:
Having the same issue and my workaround is to ignore the warning for the generated header in the following way:
```
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wfloat-equal"
#include "PackageConfig.h"
#pragma GCC diagnostic pop
```
[See here for more info on these pragmas.](https://gcc.gnu.org/onlinedocs/gcc/Diagnostic-Pragmas.html)

closed 2025-05-31 by sloretz