ros/dynamic_reconfigure #94: Build error with ueye package

tannernray, 2017-11-01:
Trying to build package ueye, throw error in trying to import SensorLevels from dynamic_reconfigure

> Base path: /home/pi/devel/james_ws
Source space: /home/pi/devel/james_ws/src
Build space: /home/pi/devel/james_ws/build
Devel space: /home/pi/devel/james_ws/devel
Install space: /home/pi/devel/james_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/pi/devel/james_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/pi/devel/james_ws/build"
####
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_py
[  3%] Built target class_loader
[  3%] Built target rosgraph_msgs_generate_messages_lisp
[  3%] Built target roscpp_generate_messages_lisp
[  3%] Built target geometry_msgs_generate_messages_py
[  3%] Built target sensor_msgs_generate_messages_py
[  3%] Built target sensor_msgs_generate_messages_cpp
[  3%] Built target geometry_msgs_generate_messages_lisp
[  3%] Built target sensor_msgs_generate_messages_lisp
[  3%] Built target geometry_msgs_generate_messages_cpp
[  3%] Built target rosgraph_msgs_generate_messages_py
[  3%] Built target roscpp_generate_messages_cpp
[  3%] Built target roscpp_generate_messages_py
[  3%] Built target rosgraph_msgs_generate_messages_cpp
[  3%] Built target _nodelet_generate_messages_check_deps_NodeletLoad
[  3%] Built target _nodelet_generate_messages_check_deps_NodeletUnload
[  3%] Built target _nodelet_generate_messages_check_deps_NodeletList
[  3%] Built target _dynamic_reconfigure_generate_messages_check_deps_ConfigDescription
[  3%] Built target _dynamic_reconfigure_generate_messages_check_deps_Reconfigure
[  3%] Built target _dynamic_reconfigure_generate_messages_check_deps_GroupState
[  3%] Built target _dynamic_reconfigure_generate_messages_check_deps_DoubleParameter
[  3%] Built target _dynamic_reconfigure_generate_messages_check_deps_ParamDescription
[  3%] Built target _dynamic_reconfigure_generate_messages_check_deps_BoolParameter
[  3%] Built target _dynamic_reconfigure_generate_messages_check_deps_Config
[  3%] Built target _dynamic_reconfigure_generate_messages_check_deps_IntParameter
[  3%] Built target _dynamic_reconfigure_generate_messages_check_deps_SensorLevels
[  3%] Built target dynamic_reconfigure_gencfg
[  6%] Built target dynamic_reconfigure_config_init_mutex
[  6%] Built target _dynamic_reconfigure_generate_messages_check_deps_Group
[  6%] Built target _dynamic_reconfigure_generate_messages_check_deps_StrParameter
[  7%] Built target gtest
[ 13%] Built target image_transport
[ 14%] Built target nodelet_topic_tools_gencfg
[ 14%] Built target bond_generate_messages_cpp
[ 14%] Built target bond_generate_messages_lisp
[ 14%] Built target bond_generate_messages_py
Scanning dependencies of target plugin_tool
[ 14%] Generating dynamic reconfigure files from cfg/stereo.cfg: /home/pi/devel/james_ws/devel/include/ueye/stereoConfig.h /home/pi/devel/james_ws/devel/lib/python2.7/dist-packages/ueye/cfg/stereoConfig.py
[ 16%] Generating dynamic reconfigure files from cfg/mono.cfg: /home/pi/devel/james_ws/devel/include/ueye/monoConfig.h /home/pi/devel/james_ws/devel/lib/python2.7/dist-packages/ueye/cfg/monoConfig.py
[ 16%] Built target _polled_camera_generate_messages_check_deps_GetPolledImage
[ 16%] Building CXX object pluginlib/CMakeFiles/plugin_tool.dir/src/plugin_tool.cpp.o
Scanning dependencies of target camera_calibration_parsers
Traceback (most recent call last):
  File "/home/pi/devel/james_ws/src/ueye/cfg/mono.cfg", line 6, in <module>
    from dynamic_reconfigure.msg import SensorLevels
ImportError: No module named msg
ueye/CMakeFiles/ueye_gencfg.dir/build.make:65: recipe for target '/home/pi/devel/james_ws/devel/include/ueye/monoConfig.h' failed
make[2]: *** [/home/pi/devel/james_ws/devel/include/ueye/monoConfig.h] Error 1
make[2]: *** Waiting for unfinished jobs....
Traceback (most recent call last):
  File "/home/pi/devel/james_ws/src/ueye/cfg/stereo.cfg", line 6, in <module>
    from dynamic_reconfigure.msg import SensorLevels
ImportError: No module named msg
ueye/CMakeFiles/ueye_gencfg.dir/build.make:83: recipe for target '/home/pi/devel/james_ws/devel/include/ueye/stereoConfig.h' failed
make[2]: *** [/home/pi/devel/james_ws/devel/include/ueye/stereoConfig.h] Error 1
[... 170 lines cut ...]
[ 17%] Built target plugin_tool
[ 20%] Built target camera_calibration_parsers
[ 23%] Built target nodelet_generate_messages_cpp
Traceback (most recent call last):
  File "/home/pi/devel/james_ws/src/ueye/cfg/stereo.cfg", line 6, in <module>
    from dynamic_reconfigure.msg import SensorLevels
ImportError: No module named msg
Traceback (most recent call last):
  File "/home/pi/devel/james_ws/src/ueye/cfg/mono.cfg", line 6, in <module>
    from dynamic_reconfigure.msg import SensorLevels
ImportError: No module named msg
ueye/CMakeFiles/ueye_gencfg.dir/build.make:83: recipe for target '/home/pi/devel/james_ws/devel/include/ueye/stereoConfig.h' failed
make[2]: *** [/home/pi/devel/james_ws/devel/include/ueye/stereoConfig.h] Error 1
make[2]: *** Waiting for unfinished jobs....
ueye/CMakeFiles/ueye_gencfg.dir/build.make:65: recipe for target '/home/pi/devel/james_ws/devel/include/ueye/monoConfig.h' failed
make[2]: *** [/home/pi/devel/james_ws/devel/include/ueye/monoConfig.h] Error 1
CMakeFiles/Makefile2:6580: recipe for target 'ueye/CMakeFiles/ueye_gencfg.dir/all' failed
make[1]: *** [ueye/CMakeFiles/ueye_gencfg.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 25%] Built target nodelet_generate_messages_py
[ 28%] Built target nodelet_generate_messages_lisp
[ 37%] Built target dynamic_reconfigure_generate_messages_lisp
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

mikaelarguedas, 2017-11-01:
Is this 3 times the same console output ? Or is there a difference between them?
It looks like you're building on a pi. can you specify your OS, architecture and ROS Distribution? How did you install dynamic_reconfigure (from source or from binaries)? 
Can you find the dynamic_reconfigure python module? (make sure to have dynamic_reconfigure installed and you're workspace sourced before running the following snippet)
```
$ python
>>> import dynamic_reconfigure
>>> from dynamic_reconfigure import msg
>>> from dynamic_reconfigure.msg import SensorLevels
```

tannernray, 2017-11-01:
Yes it was duplicate, I fixed it.

> PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
> NAME="Raspbian GNU/Linux"
> VERSION_ID="8"
> VERSION="8 (jessie)"
> ID=raspbian
> ID_LIKE=debian
> Rosversion: Indigo

Installed from source

ran snippet

> #!/usr/bin/env python
> 
> import dynamic_reconfigure
> from dynamic_reconfigure import msg
> from dynamic_reconfigure.msg import SensorLevels
> 
> print(SensorLevels)

Output:

> <class 'dynamic_reconfigure.msg._SensorLevels.SensorLevels'>

mikaelarguedas, 2017-11-01:
hmm that's interesting so it looks like the package is properly installed and on your pythonpath.
I tried in a container (ubuntu though) and could generate and compile the generated dynamic_reconfigure files successfully (console output below).

Can you try removing ueye from your workspace and build it in an overlay workspace? (to get the full cmake and build output for only this package):
```
rm -r /home/pi/devel/james_ws/src/ueye
mkdir -p /home/pi/devel/ueye_ws/src && cd /home/pi/devel/ueye_ws/src
hg clone https://bitbucket.org/kmhallen/ueye
source /home/pi/devel/james_ws/install/setup.bash
catkin_make_isolated
```

One thing to note is that ueye never built successfully on Debian Jessie on ARM architecture and has build blacklisted from the ROS buildfarm: https://github.com/ros-infrastructure/ros_buildfarm_config/pull/39/files
Maybe this question should be moved to the ueye repository and ask @kmhallen that may have more details as of why this is failing. 

Console output of a successful build of ueye:
```
root@3f735dc14bf2:~/ueye_ws# catkin_make_isolated 
Base path: /root/ueye_ws
Source space: /root/ueye_ws/src
Build space: /root/ueye_ws/build_isolated
Devel space: /root/ueye_ws/devel_isolated
Install space: /root/ueye_ws/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 1 packages in topological order:
~~  - ueye
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
The packages or cmake arguments have changed, forcing cmake invocation

==> Processing catkin package: 'ueye'
==> Creating build directory: 'build_isolated/ueye'
==> cmake /root/ueye_ws/src/ueye -DCATKIN_DEVEL_PREFIX=/root/ueye_ws/devel_isolated/ueye -DCMAKE_INSTALL_PREFIX=/root/ueye_ws/install_isolated -G Unix Makefiles in '/root/ueye_ws/build_isolated/ueye'
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /root/ueye_ws/devel_isolated/ueye
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /root/ueye_ws/build_isolated/ueye/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
-- Looking for C++ include ueye.h
-- Looking for C++ include ueye.h - not found
-- uEye SDK not found. Using downloaded files for 'amd64' architecture. (~2MB)
-- This method is only intended for the buildfarm, and will fail to communicate if the official uEye SDK is not installed.
-- The official uEye SDK can be found at https://en.ids-imaging.com/download-ueye.html (~70MB)
-- The official uEye SDK for amd64: https://en.ids-imaging.com/download-ueye.html?file=tl_files/downloads/uEye_SDK/driver/uEye_Linux_4.60_64_Bit.zip (~71MB)
[... 112 lines cut ...]
[ 16%] Built target ueye_gencfg
Scanning dependencies of target ueye_nodelets
[ 25%] [ 41%] [ 41%] [ 50%] [ 58%] Building CXX object CMakeFiles/ueye_nodelets.dir/src/Camera.cpp.o
[ 66%] [ 75%] Building CXX object CMakeFiles/ueye_nodelets.dir/src/camera_nodelet.cpp.o
Building CXX object CMakeFiles/ueye_nodelets.dir/src/CameraNode.cpp.o
Building CXX object CMakeFiles/ueye_nodelets.dir/src/framerate_nodelet.cpp.o
Building CXX object CMakeFiles/ueye_nodelets.dir/src/StereoNode.cpp.o
Building CXX object CMakeFiles/ueye_nodelets.dir/src/stereo_nodelet.cpp.o
Building CXX object CMakeFiles/ueye_nodelets.dir/src/FramerateNode.cpp.o
Linking CXX shared library /root/ueye_ws/devel_isolated/ueye/lib/libueye_nodelets.so
[ 75%] Built target ueye_nodelets
Scanning dependencies of target camera
Scanning dependencies of target framerate
Scanning dependencies of target stereo
[ 83%] [ 91%] Building CXX object CMakeFiles/framerate.dir/src/framerate_node.cpp.o
[100%] Building CXX object CMakeFiles/stereo.dir/src/stereo_node.cpp.o
Building CXX object CMakeFiles/camera.dir/src/camera_node.cpp.o
Linking CXX executable /root/ueye_ws/devel_isolated/ueye/lib/ueye/framerate
[100%] Built target framerate
Linking CXX executable /root/ueye_ws/devel_isolated/ueye/lib/ueye/camera
Linking CXX executable /root/ueye_ws/devel_isolated/ueye/lib/ueye/stereo
[100%] Built target camera
[100%] Built target stereo
<== Finished processing package [1 of 1]: 'ueye'
```

tannernray, 2017-11-02:
After running that python script I tried to build the workspace again, and I could generated and compiled the dynamic_reconfigure files successfully. So I'm not sure what changed...weird.

mikaelarguedas, 2017-11-02:
hmmm must be that something was wrong/weird in your environment, maybe your workspace was not properly sourced when you tried to build the first time. The python script you ran was only checking that the package was importable and didnt modify your environment.

Well happy ending then, I'm going to close this. Feel free to open another ticket if you face any other issue with dynamic_reconfigure

closed 2017-11-02 by mikaelarguedas