ros/dynamic_reconfigure #95: Parameter with slash isn't supported

deanyanqing, 2017-11-04:
Cfg generator doesn't support below scenario.For example, there are ros parameter as below
/node/camera/xxx
/node/lidar/xxx
Current generator try to generate c++ variable named as camera/xxx, it definitely failed.

mikaelarguedas, 2017-11-07:
Thanks @deanyanqing for your interest in `dynamic_reconfigure`.
Yes `dynamic_reconfigure` supports only parameters of primitive types (no arrays) for parameters having a name following the ROS package naming conventions "a ROS package cannot have special characters other than an underscore, and they must start with an alphabetical character".

>Current generator try to generate c++ variable named as camera/xxx, it definitely failed.

This should not happen, the generator [checks this condition](https://github.com/ros/dynamic_reconfigure/blob/17e1d4bf6a538d2def097a7f70b139ccd41e9f78/src/dynamic_reconfigure/parameter_generator_catkin.py#L74-L76) before generating the code and raises an exception with an appropriate error message. Can you give more information about what ROS distribution and dynamic_reconfigure version you are using ?

---
If you consider providing a pull request to add support for slashes, I would consider it for a future ROS release (ROS Melodic).

deanyanqing, 2017-11-08:
Yes, you are right. I put nested parameter into "group"  blindly. As you mentioned  dynamic_reconfigure not supported slash yet. But since ros already introduce yaml, there are quite a lot nested parameter as  /node/camera/xxx. It's better to support such nested parameter.What do you think?

mikaelarguedas, 2017-11-11:
>  It's better to support such nested parameter.What do you think?

I agree with you, as pointed out in my previous comment, if you provide a pull request to add support for slashes in python and C++ I would consider releasing it in the next ROS distribution.

deanyanqing, 2017-11-13:
No problem.I will submit pull request as long as get it done.

krixkrix, 2018-05-23:
To me it seems that at least the python part works if only the generator if modified.

if the [check in the generator](https://github.com/ros/dynamic_reconfigure/blob/17e1d4bf6a538d2def097a7f70b139ccd41e9f78/src/dynamic_reconfigure/parameter_generator_catkin.py#L74-L74
) is changed to 

`pattern = r'^[a-zA-Z][a-zA-Z0-9/_]*$'`     # forward slash inserted as legal character

... then the python client/server seems to work out of the box. At least I can list and retrieve parameters with slashes through the commandline tools:

`rosrun dynamic_reconfigure dynparam ...`

... and the keys (e.g. xxx/mykey) can be updated, also on the parameter server.

closed 2025-05-31 by sloretz