ros/dynamic_reconfigure #101: C++ dynamic reconfigure server with str_t param ignores numerical strings

lucasw, 2018-03-02:
https://github.com/lucasw/dynamic_reconfigure_tools/tree/master/dynamic_reconfigure_example is an example C++ server with a str_t param that it prints out when reconfigured:

```
...
gen.add("str_param", str_t, 1, "string", "")
...
```

Alphanumeric strings work as expected:
```
$ rosrun dynamic_reconfigure_example example_server_node _str_param:=5a
[ INFO] [/example_server] :[233] [Initializing nodelet with 8 worker threads.]
[ INFO] [/example_server] :[56] [5a]
```

```
$ rosrun dynamic_reconfigure_example example_server_node _str_param:=55
[ INFO] [/example_server] [/tmp/binarydeb/ros-kinetic-nodelet-1.9.14/src/loader.cpp]:[233] [Initializing nodelet with 8 worker threads.]
[ INFO] [/example_server] [/home/lucasw/ros/dr_catkin_ws/src/dynamic_reconfigure_tools/dynamic_reconfigure_example/src/example_server.cpp]:[56] []
```

The server also happens to be a nodelet, I don't think that is related but I'll create a non-nodelet example server later.

The same thing works fine with a python server (the server prints out the entire config on reconfigure):

```
$ rosrun rqt_dr_single example_server.py _str_param:="12345"
[INFO] [/rqt_dr_single_13663_1520016082376] :[15] [Reconfigure Request: 50, 0.5,\ 
          12345, True, 1]
```

lucasw, 2018-03-03:
It looks like this is a problem with C++ getParam and nothing to do with dynamic reconfigure.

lucasw, 2018-03-04:
My current workaround is to use an int type for the dynamic reconfigure value and converting it back to a string inside the reconfigure callback, though this risks a inexact string match in the case of leading zeros (fortunately I'm not dealing with floats).

mikaelarguedas, 2018-03-04:
Glad you found a workaround.

Thanks for providing the link to the upstream issue :+1: , that'll be helpful for people running into the some issue and landing on this ticket.