ros/dynamic_reconfigure #103: rostest times out on Debian Stretch and recent ubuntus

mikaelarguedas, 2018-04-09:
This seem to be Ubuntu Distro related and not ROS Distro related, consistently fails on Debian Stretch, Ubuntu Artful and Bionic.
Example jobs: http://build.ros.org/view/Mpr_ds/job/Mpr_ds__dynamic_reconfigure__debian_stretch_amd64/1/
http://build.ros.org/view/Mpr_ds/job/Mpr__dynamic_reconfigure__ubuntu_bionic_amd64/1/

krixkrix, 2018-08-06:
On debian stretch this test fails for me as well:

> rostest dynamic_reconfigure test_cpp_simple_client.launch

Or to be accurate: all tests passes but the test itself times out.

If I change this line in test_client.cpp:

`ros::AsyncSpinner spinner(2);`

To:

`ros::AsyncSpinner spinner(1);`

...then the tests succeeds.

mikaelarguedas, 2018-09-14:
thanks @krixkrix for looking into it!

I opened https://github.com/ros/dynamic_reconfigure/pull/119 that should address the test failure without modifying the number of threads used by the spinner.

PR #119 (2018-09-14): call shutdown to prevent test from hanging
Fixes #103
closed 2018-09-14 by mikaelarguedas