ros/dynamic_reconfigure #104: Python3 compatibility

vytasrgl, 2018-04-09:
Current python version is really close to be python3 compatible. Similar to roscomm i think would be great to have it compatible for py3

mikaelarguedas, 2018-04-09:
Thanks @vytasrgl for reporting it. Can you point out the parts you identified that are not python3 compatible?

vytasrgl, 2018-04-10:
I wasn't able to run full test but having fixed iteritems in: https://github.com/ros/dynamic_reconfigure/blob/17e1d4bf6a538d2def097a7f70b139ccd41e9f78/src/dynamic_reconfigure/encoding.py#L78  i was able to run reconfigure server inside python 3.6.

zchen24, 2018-04-15:
Also parameter_generator.py L180: 
                    ret.append(string.upper(x))

string.upper is not compatible

mikaelarguedas, 2018-04-16:
thank you both for the feedback!

@vytasrgl : https://github.com/ros/dynamic_reconfigure/pull/105/commits/c03c68f5736f2a9600e8a111688548ee3d653d38 should address the dictionaries issues.

@zchen24: As `parameter_generator.py` has been deprecated since the migration from rosbuild to catkin, I am not planning on making changes to it. If you know of a package that still uses it, please let me know. thanks!

mikaelarguedas, 2018-04-26:
with #105 merged I'm going to close this. Feel free to comment here or open a new issue if you find any other incompatibility.

Thanks!

closed 2018-04-26 by mikaelarguedas

Brudasoe, 2018-05-09:
[camera_aravis](https://github.com/ssafarik/camera_aravis/blob/e43cf795875ce7e162ccfad3b1ac3666e4a997f1/cfg/CameraAravisConfig.cfg#L9) still uses `parameter_generator.py`, I have the same remarks as @zchen24 - the string method causes compile errors

mikaelarguedas, 2018-05-09:
thanks @Brudasoe for following up.

I'm not opposed to update `parameter_generator.py` to be python3 compliant.
But I strongly recommend submitting Pull Requests to any package you come across that still use the deprecated rosbuild version. As that version is deprecated it is not at feature parity with the supported one and can develop bugs that have been fixed in the supported version

Brudasoe, 2018-05-10:
@mikaelarguedas do you have any pointers as to how proceed with updating this code so that it will not use `parameter_generator.py`?

mikaelarguedas, 2018-06-27:
@Brudasoe I missed that comment. The way to use the new version is to import `parameter_generator_catkin` instad of `parameter_generator` in the .cfg file. You can find an example at:

https://github.com/ros/dynamic_reconfigure/blob/a574a3b79b49d8efaf6a4a0b748b002026dd64c0/cfg/Test.cfg#L37