ros/dynamic_reconfigure #106: could not load plugin "rqt_reconfigure/Param"

mrsp, 2018-05-11:
Hi 

Any ideas on the following error when launching rqt_reconfigure:

[image]

Thx in advance

Cryptum169, 2019-03-19:
I ran into the exact same error traceback. How did you solve it? @mrsp

mrsp, 2019-03-19:
I didn't

Cryptum169, 2019-03-19:
@mrsp I think I have solved it. I was using dynamic_reconfigure for a plugin. The error came up because there's a registered dynamic reconfigure yet the node cannot be found (as it is "hiding" under another architecture of code), so the name goes empty and generates the error. I resolved the error by giving the node handle a specific name. 
`ros::NodeHandle nh;`
vs 
`ros::NodeHandle nh("~/" + name);`
If the context applies, I hope it helps.