ros/dynamic_reconfigure #115: undefined reference to `dynamic_reconfigure::__init_mutex__'

xieshuaix, 2018-08-14:
Hi, I got an error "undefined reference to `dynamic_reconfigure::__init_mutex__'" while compiling a node using dyanmic reconfigure. It seems that this has something to do with linking between TestConfig.h, config_init_mutex.h and dynamic_reconfigure_config_init_mutex.cpp when compiling the dynamic_reconfigure package. Anyone has any idea how to fix this? Thanks!

mikaelarguedas, 2018-09-14:
@xieshuaix thanks for reporting the issue. Do you happen to have a reproducible example?
Ideally a package to clone and a series of steps to build it and experience the build failure. A build log would also be helpful to know what's missing

user1409, 2018-10-22:
I am experiencing the same problem, has there been any fix so far? Have you solved the problem yourself @xieshuaix ? Would be really glad for some advise, what to try

xieshuaix, 2018-10-25:
@mikaelarguedas Thanks for your reply. I managed to get it working. What causes the problem is that the library was not properly linked.

xieshuaix, 2018-10-25:
@user1409 I fixed the issue. You could try double checking if the library is linked or not (i.e. if using cmake, find_package(dynamic_reconfigure)).

user1409, 2018-10-25:
@xieshuaix thank you for your answer. I already triple checked it. Am I missing something?

Added it here:
find_package(catkin REQUIRED COMPONENTS
  roscpp
  std_msgs
  sensor_msgs
  dynamic_reconfigure
  message_generation
        )

Put in my cfg-File at "generate_dynamic_reconfigure_options"

And made sure it is build first:
add_dependencies(my_pkg ${PROJECT_NAME}_gencfg)

Still got this error. I have a different project, where I did the same, but its just compiling fine.

mjcarroll, 2019-09-11:
It sounds like this is resolved via correct library linking.  @user1409 if you are still experiencing this, feel free to open an additional issue.

closed 2019-09-11 by mjcarroll

dupeljan, 2020-05-07:
Same error so far. My CMakeLists.txt:

cmake_minimum_required(VERSION 3.1.0 FATAL_ERROR)
project(CartConrolPlugin)

# Generate code from our own custom messages
# find_package catkin and genmsg, required for message generation macros
find_package(catkin REQUIRED COMPONENTS dynamic_reconfigure)
find_package(genmsg REQUIRED)
# Enumerate our custom messages files
add_message_files(DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR} FILES msg/Velocity.msg msg/Position.msg)
generate_messages(DEPENDENCIES)
# Generate dynamic reconf
generate_dynamic_reconfigure_options(
  cfg/PIDconf.cfg
  #...
)

#find ros
find_package(roscpp REQUIRED)
find_package(std_msgs REQUIRED)
include_directories(${roscpp_INCLUDE_DIRS})
include_directories(${std_msgs_INCLUDE_DIRS})

#find gazebo
find_package(gazebo REQUIRED)
include_directories(${GAZEBO_INCLUDE_DIRS})
link_directories(${GAZEBO_LIBRARY_DIRS})
list(APPEND CMAKE_CXX_FLAGS "${GAZEBO_CXX_FLAGS}")

#find qt
set(CMAKE_INCLUDE_CURRENT_DIR ON)
set(CMAKE_AUTOMOC ON)
set(CMAKE_AUTOUIC ON)
set(CMAKE_AUTORCC ON)
find_package(Qt5 REQUIRED COMPONENTS Widgets)

#build libruary
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} ${GAZEBO_CXX_FLAGS} ")
add_library(main SHARED main.cc)
add_dependencies(main ${PROJECT_NAME}_generate_messages)
target_link_libraries(main ${GAZEBO_LIBRARIES} ${Boost_LIBRARIES} ${roscpp_LIBRARIES})

set(CMAKE_BUILD_TYPE Debug)
#build client
#add_executable(client client.cc)
add_executable(client 
		client/main.cpp 
		client/mainwindow.cpp 
		client/mainwindow.h 
		client/mainwindow.ui 
		client/resources.qrc
		client/cartcontrollerwidget.cpp
		client/cartcontrollerwidget.h
                client/rospublisher.cpp
                client/rospublisher.h
                client/cartkinematic.cpp
                client/cartkinematic.h
	)
target_link_libraries(client ${roscpp_LIBRARIES} Qt5::Widgets)
add_dependencies(client ${PROJECT_NAME}_generate_messages)

# Build cfg server
add_executable(drfServer  drcfServer.cpp)
add_dependencies(drfServer ${PROJECT_NAME}_generate_messages ${PROJECT_NAME}_gencfg)
target_link_libraries(drfServer ${Boost_LIBRARIES} ${roscpp_LIBRARIES})

dupeljan, 2020-05-07:
According to [this question](https://answers.ros.org/question/252008/dynamic_reconfigure-tutorial-problems/), I found solution. 

As it was used [here](https://github.com/azhar92/scooter-mode/blob/master/CMakeLists.txt), you just need to add this line for proper compilation:

set_target_properties(dynamic_tutorials_server PROPERTIES LINKER_LANGUAGE CMAKE_py_LINK_EXECUTABLE)