ros/dynamic_reconfigure #116: mutex lock error

snakehaihai, 2018-08-25:
Hey i`m trying to run a code which calls for dynamic reconfgure. build has no issue. in previous 12.04 os has no issue. but once shift to 18.04. we encounter this error

demo_flight_control: /usr/include/boost/thread/pthread/recursive_mutex.hpp:113: void boost::recursive_mutex::lock(): Assertion `!pthread_mutex_lock(&m)' failed.
^CAborted (core dumped)

right after calling dynamic_reconfigure::Server<dji_sdk_demo::Aquila_paramConfig> server;
  
And it is somehow prevent code from running. 

May I ask how to solve it?

I`m using ubuntu 18.04 with ros melodic

mikaelarguedas, 2018-09-14:
@snakehaihai can you please provide a reproducible example? this issue description is unfortunately not sufficient to investigate the issue.

Thanks!

atyshka, 2019-10-29:
I too have this issue, following some code on ROS answers. My class has a private member of a `boost::recursive_mutex` and here's my constructor:

`LabelingNode::LabelingNode(ros::NodeHandle& n, ros::NodeHandle& pn)
  : config_server_(config_mutex_)
{
  boost::recursive_mutex::scoped_lock dyn_reconf_lock(config_mutex_);
  dyn_reconf_lock.unlock();
  config_server_.setCallback(boost::bind(&LabelingNode::recvConfig, this, _1, _2));
}`

My node immediately crashed with the following error:
/usr/include/boost/thread/pthread/recursive_mutex.hpp:113: void boost::recursive_mutex::lock(): Assertion `!pthread_mutex_lock(&m)' failed.
Aborted (core dumped)