ros/dynamic_reconfigure #133: C++ callback arguments

ben2789, 2019-02-19:
I am trying to use dynamic configure to set some camera parameters. I would like to pass a pointer to the camera object into the callback, so that I can use it's methods for setting the camera values. Is it possible to pass additional arguments to the dynamic reconfigure callback?
Callback:
```
void dr_callback(camera_node::dynamic_reconfig &parameters, uint32_t levels){
// set global object of parameters
}
```

Binding:
```
dr_callback = boost::bind(dr_callback, _1, _2);
dr_server.setCallback(dr_callback);
```

ideal callback:
```
void dr_callback_function(jai_camera_node::dynamic_reconfigConfig &parameters, uint32_t levels, cam_type *cam){
// use camera member functions to change values
}
```

ben2789, 2019-02-20:
using bind properly makes this not an issue.

closed 2019-02-20 by ben2789

Gurunatraj, 2022-05-30:
Hi @ben2789 , Could you please share more details on how you used the bind here...