ros/dynamic_reconfigure #150: Misorder of SYSTEM/BEFORE option in include_directories in cmake/dynamic_reconfigure-macros.cmake

hexonxons, 2020-03-19:
There is an error introduced in commit https://github.com/ros/dynamic_reconfigure/commit/c70637975dc7cd8f765efb6bd8fd57b1f0a75d05 at line https://github.com/ros/dynamic_reconfigure/blob/c70637975dc7cd8f765efb6bd8fd57b1f0a75d05/cmake/dynamic_reconfigure-macros.cmake#L118

According to [include_directories](https://cmake.org/cmake/help/latest/command/include_directories.html#command:include_directories) documentation `[SYSTEM]` options must follow `[AFTER|BEFORE]` options.
Due to this I have nice error:
`cc1plus: error: /home/hexonxons/some/path/project_name/BEFORE: No such file or directory [-Werror=missing-include-dirs]`

hexonxons, 2020-03-19:
@mjcarroll Could you possibly review and release a fixed version to melodic asap?

mjcarroll, 2020-03-19:
Good catch, thanks for the patch.

PR #151 (2020-03-19): Set right order of SYSTEM/BEFORE options in dynamic_reconfigure-macros.cmake
Fixes #150
commit 869c5bc4 (2020-03-19): Set right order of SYSTEM/BEFORE options in dynamic_reconfigure-macros.cmake. Fix #150 (#151)
closed 2020-03-19 by mjcarroll