ros/dynamic_reconfigure #162: 'ImportError: No module named client' while importing dynamic_reconfigure.client

iamrajee, 2020-08-26:
I'm trying to import dynamic_reconfigure.client in python as below
```
#!/usr/bin/env python
import roslib;roslib.load_manifest('dynamic_reconfigure')
import rospy
import dynamic_reconfigure.client //This gives _No module named client error_
```
But I getting `ImportError: No module named client`. If I only import `dynamic_reconfigure` and try to call `dynamic_reconfigure.client` then I'm getting `AttributeError: 'module' object has no attribute 'client'` error.

I've also added dynamic_reconfigure in CMakeList as
```
find_package(catkin REQUIRED COMPONENTS
  dynamic_reconfigure
)
```
My system information is 
**OS**: Ubuntu 18.04
**ROS**: Melodic

Let me know if there is anything else you would want to know. Thanks!

Gowresh-MARC, 2021-08-18:
Hello,

Was this issue solved? I a m getting the same error and i am using melodic in ubuntu 18.04 as well. 

I tried the same script in noetic in Ubuntu 20.04 and it works perfectly there.

iamrajee, 2021-08-18:
@Gowresh-MARC No, it wasn't resolved. I tried the following ways to import clients in my[ code](https://github.com/iamrajee/slam_rosmelodic_ws/blob/master/src/test_slam/scripts/dynamic_reconfigure.py):
```
import dynamic_reconfigure
# from dynamic_reconfigure import client
# from dynamic_reconfigure import *
# import dynamic_reconfigure.client
# from dynamic_reconfigure.client import Client
# from dynamic_reconfigure.dynamic_reconfigure.server import Server
```
But it dint helped. Thanks!

MatthijsBurgh, 2021-08-18:
No need to import roslib and load the manifest.

I think your PYTHONPATH is incorrect. So print that, best inside your script

iamrajee, 2021-08-18:
> No need to import roslib and load the manifest.
> 
> I think your PYTHONPATH is incorrect. So print that, best inside your script

++

Gowresh-MARC, 2021-08-19:
> No need to import roslib and load the manifest.
> 
> I think your PYTHONPATH is incorrect. So print that, best inside your script

These are my PYTHONPATHS:-

`['/home/marc/linorobot_ws/devel/lib/python2.7/dist-packages', '/opt/ros/melodic/lib/python2.7/dist-packages']
`

MatthijsBurgh, 2021-08-19:
Do you have any version of `dynamic_reconfigure` in your own ws? Or only the system installed version?

To be sure, try the following
```
PYTHONPATH=/opt/ros/melodic/lib/python2.7/dist-packages python -c "import dynamic_reconfigure.client"
```

closed 2025-05-31 by sloretz