ros/dynamic_reconfigure #181: Include order messed up

MatthijsBurgh, 2021-08-18:
I had a custom version of a pkg, including `dynamic_reconfigure` configs, in my workspace. Though I also had the released apt version installed on my machine. The customized configs had additional attributes compared to the system version. During compilation it would raise errors of missing attributes.

In the `CMakeLists.txt`, the `catkin_INCLUDE_DIRS` were added as normal includes. `/opt/ros/ROS_DISTRO/include` is part of the `catkin_INCLUDE_DIRS`.
The include dir of the generated config headers, `ws/devel/.private/PKG_NAME/include` is added as a SYSTEM BEFORE include, which I think is correct behaviour. Though this results in `/opt/ros/ROS_DISTRO/include`, which has the headers of system version of the pkg, getting priority over the include dir of the generated config headers, as all system includes come after all other includes. 

So the only solution for this and to fix the correct overlaying of workspaces, is to always include `catkin_INCLUDE_DIRS` as a system include. This way the SYSTEM BEFORE include of the generated config headers will take priority over `catkin_INCLUDE_DIRS`.
I think this should be applied in all ROS/CATKIN pkgs, but especially in pkgs using `dynamic_reconfigure`. So this doesn't require any change to the code, but I think this should be documented somewhere correctly.

So please add this to the documentation in a striking place.

MatthijsBurgh, 2021-09-04:
Fiendly but serious ping ;)

MatthijsBurgh, 2021-09-04:
I think this is related to #140 #182.

I think the implementation before #182 was correct, but required some additional documentation.

MatthijsBurgh, 2021-10-05:
@mjcarroll I do think that #182 is no the correct way to go. It is indeed the easier solution for the user, but still not correct IMO.

So either revert #182 and add documentation on this issue or release the current version. As by just merging the PR the problem is not solved, as most people use the released version.

MatthijsBurgh, 2021-11-17:
@mjcarroll ping ;)

MatthijsBurgh, 2021-12-19:
@mjcarroll friendly ping ;)

knorth55, 2021-12-24:
@MatthijsBurgh 
I have build issue introduced in #140 , and solved by #182 as #152.
#140 uses `/opt/ros/` include directories even for workspace build, which causes build error.
Just reverting #182 causes the build error again, I think.

seebq, 2022-02-15:
Confirming #182 fixes this issue for us.  CC @kakaday22

gbiggs, 2022-02-19:
Thanks for the report.

closed 2022-02-19 by gbiggs