ros/dynamic_reconfigure #195: Documentation on how the message api works

ModProg, 2023-05-01:
I'm trying to implement this for a project I'm writing in rust, so I cannot use the provided libraries.

I'm currently trying to follow the python implementation, but was wondering if there is a documentation on what a node needs to do to be compatible with dynamic_reconfigure if it is not using the libraries.

mjcarroll, 2023-05-01:
I don't recall there ever being a design doc around dynamic reconfigure.  At this point, I would consider the implementation a reference implementation.

I believe as a server, at a minimum, you need:

* `~/parameter_descriptions` - A latched publisher of type `dynamic_reconfigure::ConfigDescription` (http://docs.ros.org/en/noetic/api/dynamic_reconfigure/html/msg/ConfigDescription.html) Which enumerates the list of available parameters.
* `~/parameter_updates` - A latched publisher of type `dynamic_reconfigure::Config` (http://docs.ros.org/en/noetic/api/dynamic_reconfigure/html/msg/Config.html), which publishes the current state of the parameters
* `~/set_parameters` - A service of the that takes the type generated from the `.cfg` files.

ModProg, 2023-05-03:
I just noticed that the `editor_method` field is produced by using `repr()`... can this be assumed to be json like? And is there a reason a string is used instead of an actual object in the message? 
https://github.com/ros/dynamic_reconfigure/blob/2654f228adae0848c6e9b70fcf07f890ca6a2841/src/dynamic_reconfigure/parameter_generator.py#LL269C21-L269C73

ModProg, 2023-05-24:
For anyone interested, I documented my findings here: https://github.com/ModProg/rosrust_dynamic_reconfigure#the-dynamic_reconfigure-api

closed 2025-05-31 by sloretz