ros/dynamic_reconfigure #204: [yaml.constructor.ConstructorError] From yaml.load changed to the yaml.safe_load in the update

dengle0416, 2025-05-17:
After the recent dynamic-reconfigure update, we encountered a yaml file loading issue, as follows:

Traceback (most recent call last):
  File "/opt/ros/noetic/lib/dynamic_reconfigure/dynparam", line 239, in <module>
    elif cmd == 'load':                do_load()
  File "/opt/ros/noetic/lib/dynamic_reconfigure/dynparam", line 161, in do_load
    for doc in yaml.safe_load_all(f.read()):
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 130, in load_all
    yield loader.get_data()
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 45, in get_data
    return self.construct_document(self.get_node())
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 55, in construct_document
    data = self.construct_object(node)
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 100, in construct_object
    data = constructor(self, node)
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 427, in construct_undefined
    raise ConstructorError(None, None,
yaml.constructor.ConstructorError: could not determine a constructor for the tag 'tag:yaml.org,2002:python/object/new:dynamic_reconfigure.encoding.Config'
  in "<unicode string>", line 1, column 1:
    !!python/object/new:dynamic_reco ... 

We found that in the new version, yaml.load was changed to yaml.safe_load, causing the original yaml files to not be recognized as safe by safe_load. We tried modifying the yaml.safe_load - related updates back to yaml.load,to solve the problem. **But our question is how to load yaml parameter files successfully under yaml.safe_load in the new version.** We look forward to your helpful replies.

mjcarroll, 2025-05-17:
This was introduced in https://github.com/ros/dynamic_reconfigure/pull/202 to address a potential security concern.

Do you have a reproducible example of the yaml that triggers this?  It may be that we want to revert for the last release in noetic if this introduces a regression.

CC @florcabral as the author of the patch and @sloretz as the noetic ros boss.

dengle0416, 2025-05-18:
Great to hear from you so quickly. Here's an example of importing a yaml parameter file :

This is a dynamic_reconfigure - related node launched in the launch file:
```

<node pkg="dynamic_reconfigure" type="dynparam" respawn="false" name="dynamic_reconfigure_global_costmap_inflation_layer"
              args="load move_base/global_costmap/inflation_layer $(find configuration_parameter)/params/dynamic_reconfigure/global_costmap_inflation_layer.yaml" />

```
Below is the yaml named file "global_costmap_inflation_layer.yaml" imported by the above - mentioned node upon startup.
```

!!python/object/new:dynamic_reconfigure.encoding.Config
dictitems:
  cost_scaling_factor: 4.17985
  enabled: true
  groups: !!python/object/new:dynamic_reconfigure.encoding.Config
    dictitems:
      cost_scaling_factor: 4.17985
      enabled: true
      groups: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      id: 0
      inflate_unknown: false
      inflation_radius: 0.400
      name: Default
      parameters: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      parent: 0
      state: true
      type: ''
    state: []
  inflate_unknown: false
  inflation_radius: 0.400
state: []
```

Looking forward to your reply again.

sloretz, 2025-05-20:
It looks like the bug is that `dynparam dump` outputs yaml with `!!python/object/new` values, and so `dynparam load` can no longer load those values. This seems like a bug worth fixing.

I tested locally with https://github.com/tdenewiler/node_example. Using an older version of `dynamic_reconfigure`, I can dump and then load parameters. With the latest version I cannot:

```
root@932431678f7f:/workspaces/noetic_mini# rosrun dynamic_reconfigure dynparam dump pytalker asdf.yaml
root@932431678f7f:/workspaces/noetic_mini# rosrun dynamic_reconfigure dynparam load pytalker asdf.yaml
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/dynamic_reconfigure/dynparam", line 239, in <module>
    elif cmd == 'load':                do_load()
  File "/opt/ros/noetic/lib/dynamic_reconfigure/dynparam", line 161, in do_load
    for doc in yaml.safe_load_all(f.read()):
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 130, in load_all
    yield loader.get_data()
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 45, in get_data
    return self.construct_document(self.get_node())
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 55, in construct_document
    data = self.construct_object(node)
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 100, in construct_object
    data = constructor(self, node)
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 427, in construct_undefined
    raise ConstructorError(None, None,
yaml.constructor.ConstructorError: could not determine a constructor for the tag 'tag:yaml.org,2002:python/object/new:dynamic_reconfigure.encoding.Config'
  in "<unicode string>", line 1, column 1:
    !!python/object/new:dynamic_reco ... 
    ^
```

> But our question is how to load yaml parameter files successfully under yaml.safe_load in the new version

You'll have to remove all uses of `!!python/object/new`.

Before:

```
!!python/object/new:dynamic_reconfigure.encoding.Config
dictitems:
  a: 1
  b: 2
  enable: true
  groups: !!python/object/new:dynamic_reconfigure.encoding.Config
    dictitems:
      a: 1
      b: 2
      enable: true
      groups: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      id: 0
      message: hello
      name: Default
      parameters: !!python/object/new:dynamic_reconfigure.encoding.Config
        state: []
      parent: 0
      state: true
      type: ''
    state: []
  message: hello shane
state: []
```

After:

```
a: 1
b: 2
enable: true
message: hello shane
```

And then loading works again.

dengle0416, 2025-05-20:
Thanks for your response. While removing all the "!!python/object/new" tags from the YAML file allows it to load successfully, there are still two minor issues:

1. This modification disables the dynamic parameter adjustment capability.
2. The "dynparam dump"  still generates YAML files containing "!!python/object/new" tags, which might cause issues in future modifications.

We think reverting yaml.safe_load back to the original yaml.load in the new release might be a good approach. Looking forward to your reply again.

sloretz, 2025-05-20:
> This modification disables the dynamic parameter adjustment capability.

I'm not sure what you mean.

1. I used `dynparam dump` to dump the parameters to a yaml file
1. I modified the yaml file to remove `!!/python/object/new`
2. I changed the parameters in the yaml file
3. I used `dynparam load` to load the new yaml file
4. I confirmed the parameters were updated in rqt.
5. I used rqt to modify the parameters dynamically
6. I used `dynparam dump` to dump the parameters, they showed the changes made by rqt

From those steps, it looks like the workaround of editing out `!!python/object/new` did not affect the ability to adjust the parameters dynamically.

---

> We think reverting yaml.safe_load back to the original yaml.load in the new release might be a good approach

I think either reverting, or modifying `dynparam dump` to not output `!!python/object/new` could be good solutions. I'm going to look closer at the linked CVE to try to decide which way we should go.

sloretz, 2025-05-20:
Looking into https://github.com/advisories/GHSA-92fv-mr4v-6fw4 I think the CVE score of 8.4/10 is **incorrect**. If I understand correctly, the score should be https://www.first.org/cvss/calculator/3-1#CVSS:3.1/AV:L/AC:L/PR:L/UI:R/S:U/C:H/I:H/A:H  `7.8/10`. A comparable CVE is https://nvd.nist.gov/vuln/detail/cve-2024-39383

User interaction `None` -> should be `Required`. You can construct a `malicious.yaml` and run `dynparam load malicious.yaml` on your machine all you want. You can also run `python malicious.py` on your machine all you want. Neither of those are attacks. If you want to do something malicious with this you need to trick someone else into running `dynparam load malicious.yaml` on their machine.

---

How serious of a problem is this in practice? I'm not sure. If it's easy to make `dynparam dump` output yaml without arbitrary Python objects then I'll go ahead and do that. If it's not, I'll revert #202

sloretz, 2025-05-20:
I have a fix in #205 that uses a custom loader and dumper to represent the Config objects with `!dynamic_reconfigure.encoding.Config` tags. However, I can't be sure that this doesn't break something else. There's not enough time to robustly test this fix. Since the CVE requires user interaction, I'm going to revert #202.

PR #206 (2025-05-20): Revert "Fix unsafe yaml load on dynparam (#202)"
This reverts commit 9975cc8b55b3039115da6662cc7279cc65303844.

@florcabral FYI 

#202
Fixes #204
Alternative to #205
closed 2025-05-20 by sloretz